PetoiCamp/OpenCat-Quadruped-Robot #2: OpenCat.ino Sketch to big to compile

loeweh, 2021-02-22:
Hi,

downloaded OpenCat.ino. It does not compile. Error is:

Sketch uses 33764 bytes (104%) of program storage space. Maximum is 32256 bytes.
Global variables use 1344 bytes (65%) of dynamic memory, leaving 704 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.

Updated libraries to the latest.

Any idea?

BR Heiko

borntoleave, 2021-02-22:
You need to download version 2.6.1 of the IRremote library and set the
unused encoders to 0s.
More detailed instructions are mentioned in chapter 4 of the user manual at
bittle.petoi.com.

regularfry, 2021-02-24:
Can we get this documented in the Nybble guide too, please?  This affects the v0_2 board I've got too.

borntoleave, 2021-02-25:
Yes, it's already mentioned in the manual: https://nybble.petoi.com/chapter4#4-2-1-install-through-the-library-manager

ukkuroda, 2021-03-14:
For IR remote comes with biddle,
 only "DECODE_NEC", "DECODE_HASH" set to 1 and others set to 0.
You can find it in the image.
The document says "set unused decoders to 0" but there's no way to tell that the remote sends NEC codes.

borntoleave, 2021-03-15:
In the manual, there's a screenshot showing only the "DECODE_NEC" and
"DECODE_HASH" are set to 1, and others set to 0.
I'll add a sentence to explicitly indicate that.

closed 2021-03-24 by borntoleave

kkiyama117, 2021-09-24:
Same error occured in my environment with the latest version of opencat.
Sketch uses 32448 bytes and global variables uses 1208 bytes.
What can I do to solve this issue?

borntoleave, 2021-09-24:
When did you download the code?
5 days ago I included the modified IR library in the OpenCat folder. You no
longer need to modify any files to get the code compiled and the flash
usage will be about 89%.

kkiyama117, 2021-09-24:
I downloaded opencat yesterday, and do as　written in [docs](https://bittle.petoi.com/4-configuration).
 I haven't downloaded IRremote and also edited any files in opencat folder
Please tell me where I should check, because maybe I'm wrong. It seems 
writeinstinct.ino works well.

borntoleave, 2021-09-24:
Did you start from a clean Arduino installation? Do you have some older
libraries installed previously?

kkiyama117, 2021-09-24:
Since it's my first time using Arduino IDE, no libs should be installed but I'm going to try to reinstall everything. (I don't have time today, so I'll report back if it doesn't work.)

kkiyama117, 2021-09-25:
Reinstalling doesn't work well for me.
I'm using
- Arduino IDE 1.8.16 (installed by Pacman, installer of Manjaro Linux)
- [this version of I2Cdevlib](https://github.com/jrowberg/i2cdevlib/tree/c8e05299df44a974d0df07eb4b1a7b5893640ace)
- PWM Servo driver Library 2.4.0
- QList 0.6.7
- main branch of OpenCat

And the error occurs in compiling. I can't copy logs directly since I'm using Japanese translation, but it says 
```
Sketch uses 32448 bytes (100%) of program storage space. Maximum is 32256 bytes.
Global variables use 1208 bytes (58%) of dynamic memory, leaving 840 bytes for local variables. Maximum is 2048 bytes.
text section exceeds available space in board
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
```

borntoleave, 2021-09-25:
> Reinstalling doesn't work well for me.
> I'm using
>
> * Arduino IDE 1.8.16 (installed by Pacman, installer of Manjaro Linux)
> * [this version of [...]

Which board are you selecting? Are you choosing Arduino Uno?

kkiyama117, 2021-09-25:
I'm using version 1.0, and I choose Arduino Uno also.

kkiyama117, 2021-09-25:
I got it with using Arduino avr boards not with downloaded default by Arch Linux but with installed manually. Thanks for providing 　the beneficial support.