PetoiCamp/OpenCat-Quadruped-Robot #3: Nybble

Troy-Chard, 2021-02-28:
Not sure if this is a bug or not, but thought I'd mention and let you decide :)

Last night, while trying to write custom zero skill for Nybble, it didn't seem to work. 

The fix was easy: in my local branch: 
1. I moved zero to progmemPointer[] array.
1. Revised it to "zeroN" in skillNameWithType array.

And to gain a bit more space in Program memory I commented out hs, hs1, hs2 skills.

borntoleave, 2021-03-01:
>    2. Revised it to "zeroN" in skillNameWithType array.

It relates to the memory location to save a skill. The code posted on
GitHub should have the right settings. Do you have a screenshot showing
the codes that don't work? Was it identical to the raw GitHub code?

Troy-Chard, 2021-03-01:
It could be what's in the repo is correct as is :) 

I just noticed that, before I could use the zero skill myself to try a new behaviour, I needed to add an entry to the [progmemPointer array in InstinctNybble.h](https://github.com/PetoiCamp/OpenCat/blob/9571607df11fe6ecb07a70122fb2792cdd7a202e/WriteInstinct/InstinctNybble.h#L640).  

( I saw the [progmemPointer already has it in InstinctBittle.h](https://github.com/PetoiCamp/OpenCat/blob/9571607df11fe6ecb07a70122fb2792cdd7a202e/WriteInstinct/InstinctBittle.h#L694) )

Also I revised the name in the [skillNameWithType](https://github.com/PetoiCamp/OpenCat/blob/9571607df11fe6ecb07a70122fb2792cdd7a202e/WriteInstinct/InstinctNybble.h#L636
) array to be "ZeroN" instead of "ZeroI"  ( following the example in InstinctBittle.h ) 

( But then OpenCat.ino got too big, so I commented out hs, hs1, hs2, entries ).

After editing my skill a few times and uploading, it didn't seem to work though (it just kept doing similar to calib pose iirc).

Then I realized I needed to run WriteInstinct.ino to rewrite the EEPROM with new settings moving zero from EEPROM to PROGMEM 💡 and then my zero Newbility worked :)

The new entries in my copy here appear as: 
```
#if !defined(MAIN_SKETCH) || !defined(I2C_EEPROM)
		//if it's not the main sketch to save data or there's no external EEPROM, 
		//the list should always contain all information.
  const char* skillNameWithType[]={"bdI","biI","bkI","bkLI","bkRI","crFI","crLI","crRI","lyI","trFI","trLI","trRI","vtI","wkFI","wkLI","wkRI","balanceI","buttUpI","calibI","droppedI",/*"hsN","hs1N","hs2N",*/"liftedI","luI","restI","sitI","sleepI","strI","zeroN","hiI","peeI","puI","rcI",};
  const char* progmemPointer[] = {bd, bi, bk, bkL, bkR, crF, crL, crR, ly, trF, trL, trR, vt, wkF, wkL, wkR, balance, buttUp, calib, dropped, /*hs, hs1, hs2,*/ lifted, lu, rest, sit, sleep, str, zero, hi, pee, pu, rc, };
#else	//only need to know the pointers to newbilities, because the intuitions have been saved onto external EEPROM,
	//while the newbilities on progmem are assigned to new addresses
  const char* progmemPointer[] = {/*hs, hs1, hs2,*/ zero, };
#endif
```

( and this is just me fooling around trying my first newbility... really I should maybe be writing my own .ino. eventually :P

Troy-Chard, 2021-03-01:
( more background ... ) 

My Newbility is sad so far though...
```
const char zero[] PROGMEM = {  // turtle roll?   ( it's not even close yet :(
-5, 0, 0, 1,
 0, 0, 0,
  -10,-100,   5,   0,   0,   0,   0,   0,  80,  80, -80, -80, -55, -55,  55,  55,   8, 10, 0,0, // tuck
    0,  10,   5,   0,   0,   0,   0,   0,  84,  84,  80,  80,  90,  90, -90, -90,   2, 4,  0,0,  // arch
   70,  10,   5,   0,   0,   0,   0,   0, -90,   0,   0,  90,  90,  90, -90, -90,   9, 2,  0,0,  // roll - speed 100 should cause movement in one step, but, wow, hard on the servos :( 
  -70,  40,   5,   0,   0,   0,   0,   0,   0, -90,  90,   0,  90,  90, -90, -90,  30, 6,  0,0,  // Flip.A      
    0,   0,   0,   0,   0,   0,   0,   0,  30,  30, -30, -30,  30,  30, -30, -30,   2,20,  0,0, // stand?  
};
```

But my Nybble has protective gear to handle testing which I can peel off later pretty easy :)

[image]

[image]

borntoleave, 2021-03-01:
Oh, yes, thanks for reporting. I have added zero back to the newbility list
and also updated the data structure of Nybble behaviors for the new code.

Troy-Chard, 2021-03-01:
Thanks RZ! :)

borntoleave, 2021-03-02:
Have you disabled the unused encoders in IRremote.h? It's about 85% after compiling on my side.

Troy-Chard, 2021-03-02:
I'll try that, thanks!   

The problem I noticed with sketch size is in the current main branch of OpenCat :(  
I just don't want people trying Nybble for the first time to hit the problem.  ( sorry for the troubles my change request is causing 😮 )

borntoleave, 2021-03-02:
If you disable the unused encoders in IRremote.h, it will save about 10%
space.
I verified the compiling result on my end:
*Bittle*
Sketch uses 28374 bytes (87%) of program storage space. Maximum is 32256
bytes.
Global variables use 1198 bytes (58%) of dynamic memory, leaving 850 bytes
for local variables. Maximum is 2048 bytes.
*Nybble *
Sketch uses 28244 bytes (87%) of program storage space. Maximum is 32256
bytes.
Global variables use 1202 bytes (58%) of dynamic memory, leaving 846 bytes
for local variables. Maximum is 2048 bytes.

Troy-Chard, 2021-03-02:
Thanks borntoleave, that's great :)

closed 2021-03-02 by Troy-Chard

Troy-Chard, 2021-03-04:
Quick follow-up... 85% is what I get too.  Thanks again borntoleave :)

[image]