PetoiCamp/OpenCat-Quadruped-Robot #12: Divide by zero

DaveB1620, 2021-09-24:
File WriteInstinct/OpenCat.h line 926:
    void calibratedPWM(byte i, float angle, float speedRatio = 0) {
All active calls to calibratedPWM() only have 2 arguments, so the default value of speedRatio (0) is used unguarded on line 924:
   byte steps = byte(round(abs(duty - duty0) / 1.0/*degreeStep*/ / speedRatio)); //default speed is 1 degree per step
triggering a divide by zero.

borntoleave, 2021-09-24:
According to the later comment on line 934, it seems that my previous code
was using 1 as the default value.
speedRatio = 0 was introduced recently as the maximal speed without
interpolation.
I used the calibratedPWM() with 2 arguments in other places. I wonder why
the divide-by-zero error is not breaking the program?
[image: Screen Shot 2021-09-25 at 12.19.09 AM.png]

borntoleave, 2021-12-29:
> File WriteInstinct/OpenCat.h line 926:
> void calibratedPWM(byte i, float angle, float speedRatio = 0) {
> All active calls to calibratedPWM() only have 2 arguments, so the default value of speedRatio [...]

I fixed the problem, though it didn't trigger warnings in the program. 
```cpp
  byte steps = speedRatio ? byte(round(abs(duty - duty0) / 1.0/*degreeStep*/ / speedRatio)) : 0; 
  //if default speed is 0, no interpolation will be used
  //otherwise the speed ratio is compared to 1 degree per second.
```

closed 2021-12-29 by borntoleave