PetoiCamp/OpenCat-Quadruped-Robot #13: Understanding the format of a behavior: trigger axis

durka, 2021-12-26:
The `bf` behavior is defined as: https://github.com/PetoiCamp/OpenCat/blob/f1013cece692d6ff9d86577a7216416f9587aa2d/WriteInstinct/InstinctBittle.h#L599-L610

If I understand correctly, the fifth frame is triggered by a pitch angle of -10 degrees. The sixth frame seems to have the trigger axis set to zero (i.e. no trigger axis) but there is an angle value. It seems that the `127` will never be read because the axis is tested first and you can't trigger on yaw: https://github.com/PetoiCamp/OpenCat/blob/f1013cece692d6ff9d86577a7216416f9587aa2d/OpenCat.ino#L740-L742

Is this meant to be ignored, or an error, or do I misinterpret the format?

borntoleave, 2021-12-26:
The parameters for the bf (backflip) skill may be different for each model
due to the inconsistency of many factors, like calibration and surface
friction. As far as I tested on my end, most of the models can complete a
backflip. Since you may need to tune the parameters by yourself, I just
left a few non-functional residues during experiments as is.

durka, 2021-12-26:
OK, so indeed that 127 doesn't do anything but doesn't hurt anything by being there. Thanks for confirming!

closed 2021-12-26 by durka