PetoiCamp/OpenCat-Quadruped-Robot #16: New serial code doesn't work with RPi

sskorol, 2021-12-30:
Hi @borntoleave @JasonWong08,

I noticed a recent update in serial code. Wondering if you have tested it with an RPi hat?

It doesn't work for me. First of all, it incorrectly detects a serial port: `/dev/ttyAMA0` is wrong for RPi. There should always be `ttyS0`.

I updated it manually in code, but serial behavior still seems incorrect to me:

- Some commands work, the others - not. E.g. I can run `ksit`, but further `kbalance` doesn't work.
- NyBoard is always stuck after 1 or 2 commands and requires a reset.
- Sometimes I see response parsing exceptions for the correct commands, e.g. [here](https://github.com/PetoiCamp/OpenCat/blob/main/serialMaster/ardSerial.py#L155) when we try to extract a 0-element from None object.
- NyBoard is totally unstable: it may beep several times before or after the command's execution but there's no response at all.

Note that I can still control Bittle with IR until the board is stuck on serial.

I can record a video of course. But it's not really helpful then the board is stuck in 1 command and quickly becomes unresponsive.

Any thoughts or comments on how to make the serial interface stable with an RPi hat would be greatly appreciated.

JasonWong08, 2022-01-01:
@sskorol Thank you for your committing the issue. we have done a workaround to fix it.

borntoleave, 2022-01-04:
We have added a logic to automatically select the Pi's port when it's connected directly through the 2x5 socket (not through USB). You can still manually define the port if the automatic logic doesn't work in your case. 

We tested the script on Raspian. We didn't observe other problems you mentioned in your issue. The script can keep taking serial commands until you enter "q" or "quit". 

Please let us know if you have more problems, or is it ok to close this issue.

sskorol, 2022-01-04:
Ok, thanks, I'll test it today.

sskorol, 2022-01-05:
@borntoleave @JasonWong08 I tested the new code and still see the same issues. I recorded the video with mostly all of them: https://youtu.be/v-gfkdAKLus

sskorol, 2022-01-05:
I have a quick update. Just googled issues with RPi serial port. Found the following thread: https://raspberrypi.stackexchange.com/questions/114366/rpi4-serial-port-not-working-on-either-raspberry-os-or-ubuntu

I followed all the steps from the answer and now the serial port works as expected. I tried different commands. There are no hangings or weird movements anymore. Most likely there were some serial interruptions with integrated BT, as I noticed a `dtoverlay` in my boot config + there was a `python-serial` installed via apt.

Anyway, it's worth copy-pasting the following instructions to your official docs for all the users that may face similar issues with RPi hats:

- add `enable_uart=1` to `/boot/config.txt`
- remove `console=serial0,115200` from `/boot/firmware/cmdline.txt` on Ubuntu and `/boot/cmdline.txt` on Raspberry Pi OS
- disable the serial console: `sudo systemctl stop serial-getty@ttyS0.service && sudo systemctl disable serial-getty@ttyS0.service`
- make sure you have `pyserial` installed if you're using the python serial library, not python-serial from apt.
- create the following udev file (I created `/etc/udev/rules.d/50-tty.rules`):
```
KERNEL=="ttyS0", SYMLINK+="serial0" GROUP="tty" MODE="0660"
KERNEL=="ttyAMA0", SYMLINK+="serial1" GROUP="tty" MODE="0660"
```
- reload your udev rules: sudo udevadm control --reload-rules && sudo udevadm trigger
- change the group of the new serial devices:
```
sudo chgrp -h tty /dev/serial0
sudo chgrp -h tty /dev/serial1
```
- The devices are now under the tty group. Need to add the user to the tty group and dialout group:
```
sudo adduser $USER tty
sudo adduser $USER dialout
```
- update the permissions for group read on the devices:
```
sudo chmod g+r /dev/ttyS0
sudo chmod g+r /dev/ttyAMA0
```
- reboot

sskorol, 2022-01-05:
BTW, here's what I get from `platfrom.uname()` in Python on RPi / Ubuntu:
```
uname_result(system='Linux', node='rpi', release='5.4.0-1047-raspi', version='#52-Ubuntu SMP PREEMPT Wed Nov 24 08:16:38 UTC 2021', machine='aarch64', processor='aarch64')
```

JasonWong08, 2022-01-05:
@sskorol, Great! You got the key! you can refer to my post on OpenCat Forum about how to solve this issue (https://www.petoi.camp/forum/software/fixed-the-issue-on-github-new-serial-code-doesn-t-work-with-rpi-hat-16).
actually we use the serial port - **/dev/ttyS0**
So on your side, you can modify the script simply like this:
[image]
Please have a try again and feel free to contact us. Thanks a lot.

JasonWong08, 2022-01-05:
For raspberry Pi hard hardware, if you plug it in Nyboard,  We actually use **/dev/ttyS0**. No matter what operating system you installed (Ubuntu or Raspbian etc.), you should enable it at first. 
If you installed Raspbian, you can refer to the Section 4.4.1. (Config Raspberry Pi serial port) in the user manual
https://bittle.petoi.com/4-configuration

If you installed Ubuntu, you can refer to @sskorol 's comment above.

sskorol, 2022-01-05:
> @sskorol, Great! You got the key! you can refer to my post on OpenCat Forum about how to solve this issue [...]

@JasonWong08 @borntoleave if you take a look at the above comments and video, you'll notice that there was no problem for me to change a port in a code. The issue I highlighted is about the actual fact that `platform.uname()[1] == 'raspberrypi'` you use for automatic `ttyS0` assignment in a script doesn't work on Ubuntu/RPi (see the output of a https://github.com/PetoiCamp/OpenCat/issues/16#issuecomment-1005307333).

JasonWong08, 2022-01-05:
@sskorol, thank you for your advice. We'll fix it in a better way.

closed 2022-10-25 by borntoleave