PetoiCamp/OpenCat-Quadruped-Robot #19: Serial commands are sometimes skipped

sskorol, 2022-01-14:
**Robot**: Bittle
**Hardware**: NyBoard v1.0, RPi Zero 2 W
**Software**: Ubuntu Server 20.04, Python 3.8

**Steps**:
1. Run `ardSerial.py`
2. Send **kbalance** command
3. Send **krest** command
4. Send **kbalance** command
5. Send **krest** command

**Expected**: all the commands are successfully executed, Bittle performed the requested actions.
**Actual**: the last `krest` is executed, but Bittle didn't perform any action.

Check the [video](https://youtu.be/CsW0uFY08yg) to see it in action.

Note that it's not related to `krest` command only. I observed the same behavior for any command. Everything may work smoothly, and then in a moment, some command is just skipped.

Also note that seems like I'm not alone here. I investigated existing GitHub repos for Bittle and found [this one](https://github.com/EnriqueMoran/BittleXboxController) for the gamepad teleop.  There's a list of know issues at the end.

[image]

The second one seems very close to what I'm talking about. And this repo uses [pyBittle](https://github.com/EnriqueMoran/pyBittle) library as a dependency for Serial communication.

Anyway, it seems like there's a bug. So would be greatly appreciated if you can take a look @borntoleave @JasonWong08.

Thanks,
Sergey

borntoleave, 2022-01-16:
There seems to be a bug in the motion code when loading skills. I tried to
find it but was not successful. I'll give it another try this month.
Before that, you may try sending the command twice or send it right after
"balance"?

sskorol, 2022-01-16:
@borntoleave it'd probably work. But the main logical issue here is that each skill behaves like an infinite loop of repetitive operations. In this case, we always want to prevent redundant commands sending (aka duplicates) by checking what was the last action. Otherwise, teleop tools that are designed to constantly send the recent state would just DDOS the Bittle. And that's why in the most of implementations people add an additional protection to serial code e.g. `if current_command != previous_command`. That's what I did as well. Yeah, I can still send the command twice if I want. But the things might become much easier if apart from the skills there will be implemented a more common for ROS technic, when we move objects based on (x, y, z, th) vectors. Then I can constantly send just coordinates's updates, and Bittle would decide how to reach those points.

sskorol, 2022-01-16:
BTW, can you provide a pattern for a single "step"? E.g. when we walk or crawl, such skills consist of a number of repetitive steps I guess. I want to extract it for testing.

borntoleave, 2022-01-16:
> BTW, can you provide a pattern for a single "step"? E.g. when we walk or crawl, such skills consist of a number of repetitive steps I guess. I want to extract it for testing.

Yes, in instinctBittle.h, you can find the frames of gaits. The explanation of the format can be found at https://bittle.petoi.com/11-tutorial-on-creating-new-skills.

sskorol, 2022-01-16:
@borntoleave I checked the guide but it's not really clear what are 1-7 joint angles for in the provided table? And what would happen if I change 1-7 values while calibration?

[image]

Moreover, I see that we can set the head's angle for static postures. But it's missing for gaits. Can you elaborate on that? And how can I control it? If we talk about camera usage, I'd prefer disabling the head's rotation while moving, as RPi can't stabilize the video and streaming becomes awful.

sskorol, 2022-01-17:
Seems like I found where the head is controlled in code: https://github.com/PetoiCamp/OpenCat/blob/main/OpenCat.ino#L828-L833

But it's not really convenient that I need to disable it globally. Would be great to have some option probably to have more flexible head control.

P.S. The question regarding 1-7 joint angles is still actual.

borntoleave, 2022-01-17:
> Seems like I found where the head is controlled in code: https://github.com/PetoiCamp/OpenCat/blob/main/OpenCat.ino#L828-L833
> 
> But it's not really convenient that I need to disable it globally. Would be great to have some option probably for the more flexible head control.
> 
> P.S. The question regarding 1-7 joint angles is still actual.

The 1~7 joints are reserved for our 12 DOF robot, with head, tail, and should roll angles. 

The joints are grouped by four for the convenience of logic conditions in the code: 

|                  |                  |                  |                  |
|------------------|------------------|------------------|------------------|
| head pan,        | head tilt        | tail pan         | N/A              |
| shoulder roll 1  | shoulder roll 2  | shoulder roll 3  | shoulder roll 4  |
| shoulder pitch 1 | shoulder pitch 2 | shoulder pitch 3 | shoulder pitch 4 |
| knee pitch 1     | knee pitch 2     | knee pitch 3     | knee pitch 4     |

Gaits use periodical data. If I include the head and unused joints' angles (0-7 joints), it will double the flash memory usage and exceed the available SRAM on ATmega328P. So for the 8 DOF models, I only save 8 joints (8-11 joints) per frame for the gaits. The postures and behaviors have only one frame or a few frames, so I kept all the joints for better compatibility for my future models. Changing the calibration value of 1-7 joints won't affect anything because no servos are attached to those ports. 

The head movement is generated on the fly as a function of the period of the gait. It's a trade-off for limited flash space. The good part is that you may adjust the amplitude and period of the rotation to compensate for the wobbling of the body.

sskorol, 2022-01-17:
Ok, thanks for clarification! I'll play with head's rotation.

Let me know when you find the root cause of the original issue with serial commands please.

borntoleave, 2022-03-17:
Hi,
I have fixed the issue and pushed a big update to the firmware: https://github.com/PetoiCamp/OpenCat/tree/2.0
You can read the changelog for the major updates: https://github.com/PetoiCamp/OpenCat/blob/2.0/ChangeLog.md

sskorol, 2022-03-17:
Hi, thanks for the update. Unfortunately, I can't check it at the moment. You know what happens in Ukraine. Bittle is waiting at home until the war is over.🇺🇦

JasonWong08, 2022-03-18:
> Hi, thanks for the update. Unfortunately, I can't check it at the moment. You know what happens in Ukraine. Bittle is waiting at home until the war is over.🇺🇦

Take care bro.

borntoleave, 2022-04-14:
Fixed in the 2.0 branch

closed 2022-04-14 by borntoleave