PetoiCamp/OpenCat-Quadruped-Robot #20: Getting IMU data

sskorol, 2022-01-18:
Hi @borntoleave,

Is it possible to retrieve IMU data from Bittle programmatically? I want to get odometry for SLAM package integration.

Thanks,
Sergey

borntoleave, 2022-01-18:
You can delete the macro for DEVELOPER, and use the code blocks inside to print out the angles. The format is yaw, pitch, roll.

```c
#ifdef FIX_OVERFLOW
      for (byte g = 1; g < 3; g++) {
        yprLag[lag][g] = ypr[g];
        ypr[g] = yprLag[(lag - 1 + HISTORY) % HISTORY][g] ;
      }
      lag = (lag + 1) % HISTORY;
#endif

#ifdef DEVELOPER
      PT(ypr[0]);
      PTF("\t");
      PT(ypr[1]);
      PTF("\t");
      PTL(ypr[2]);
#endif
    }

sskorol, 2022-01-18:
@borntoleave and what about coordinates? I guess they are stored in `gravity` vector?

borntoleave, 2022-01-18:
If you look at the testMPU.ino in the ModuleTest folder, you can see how to print out the x y z values.

sskorol, 2022-01-18:
@borntoleave great, thank you!

sskorol, 2022-01-22:
@borntoleave to provide sufficient info to nav package I need to collect linear acceleration, angular velocity and orientation from IMU. Just want to confirm this info can be collected from Bittle. I see several examples with accel, but not really sure what’s the difference between common accel and world accel? I also see quaternion structure in code. Is it sufficient to determine orientation? Or I need to apply additional formulas against pitch, roll and yaw to compute it? And for angular velocity I guess I can use roll, pitch and yaw, right?

borntoleave, 2022-01-22:
The MPU6050 code is cited from this project:
https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050

The ypr from the code is yaw, pitch, roll angles, not velocities. You
always measure acceleration then integrate to get the velocity and
position.

If you plot the common accel and world accel and see their feedback with
your motion, you can compare the differences and decide the best one. I'm
using aaWorld for my experiments.

sskorol, 2022-01-25:
@borntoleave I want to try another IMU (10 DOF). However, i2c port on RPi is occupied by the NyBoard. Can I access NyBoard's i2C from RPi if I solder the sensor's pins to the available slots above the i2c groove socket?
[image]

borntoleave, 2022-01-25:
In short, yes. As long as you keep the I2C switch on Arduino, the I2C port
of RPi is dangling. You can connect external sensors to its port. You need
to keep the magnetic sensors away from the body to avoid interference by
the servos' motion because the servos emit very strong EM fluctuation.

If you dial the I2C switch from Arduino to RPi. The I2C grove and all the
devices on I2C will be taken over by the RPi. However, the servo driver
PCA9685 will also be controlled by RPi, and the motion pattern generator
ATmega328P won't be in effect.

sskorol, 2022-01-25:
> You can connect external sensors to its port.

Do you mean RPi port or those i2C port linked with a groove?
RPi port is already occupied with a hat. I just need to access an external i2C sensor via RPi.

Are you saying it's possible only if RPi switch is on?

borntoleave, 2022-01-25:
[image]
As long as the I2C switch is not dialed at RPi, the I2C port of RPi is not really connected to the board's I2C network. 
The switch can disconnect the I2C network from 328P and connect them to the RPi.

sskorol, 2022-01-25:
@borntoleave but even if it's not connected to NyBoard's i2c network, the RPi pins would still be occupied if we use Zero or A+ board as a hat. So we can't physically connect sensors directly to Pi's i2c. And if a switch the I2C network from 328P, I won't be able to control servos via OpenCat code anymore, as you have mentioned. So what would be your recommendation in this case? What's the best option of connecting i2c sensor to Pi mounted on top of 328P w/o loosing the main functionality.

borntoleave, 2022-01-25:
You can jump four wires between your sensor and the Pi: Vcc, GND, SCL, and
SDA.
Pay attention to the voltage level of Vcc, SCL, and SDA. The voltage level
of SCL and SDA on Pi is 3.3V.
For Vcc, you can choose either 3.3V or 5V from Pi to fit your sensor's
needs.

sskorol, 2022-01-25:
But SCL and SDA pins are occupied by NyBoard when Pi is used as a hat. And 2 extra SCL/SDA pins on Pi are intended to eeprom usage only. They won't work for common use cases. So how can I wire SDA/SCL between Pi (connected to NyBoard) and a sensor then?

borntoleave, 2022-01-25:
> SCL and SDA pins are occupied by NyBoard (2x5 socket) when Pi is used as a hat.

Yes.  But they are not really connected to anything devices. So you can solder two wires to the back of the pins on the Pi.

sskorol, 2022-01-29:
@borntoleave well, I did it, but here's a new issue. When I run `i2cdetect -y 1`, it gives me the following picture:

[image]

By default, if don't connect RPi to NyBoard, I have only a single 68 address occupied by my sensor. But the RPi is used as a hat, I see this weird matrix. 68 address doesn't work anymore. How can I guess which one is mine now w/o brute-forcing each address? I'm also wondering why all those addresses are occupied. Is NyBoard really bind them for something?

sskorol, 2022-01-29:
Ok, from the above picture I see that 0x68 address is used by MPU6050. So that's a problem. 2 sensors have the same address. Any thoughts on how to resolve this conflict?

sskorol, 2022-02-07:
Made it work via software i2c bus. Details here: https://forums.raspberrypi.com/viewtopic.php?p=1968858#p1967861

@borntoleave I see you committed accel data printing feature. Have you also tried to output raw gyro data? I played with it a bit based on the provided samples but always got constant values like [0;0;1]. Also, wondering why there are int values and not float like for accel?

borntoleave, 2022-02-08:
Thanks for the note on I2C. 

The IMU code is mostly from [i2cdevlib](https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/examples/MPU6050_DMP6_using_DMP_V6.12/MPU6050_DMP6_using_DMP_V6.12.ino). The raw gyro data is **VectorInt16 gy** and you may print it out similar to the example code. 

The accel data is read from the register and is in integer type. You may divide some factors to get real accelerations in float. 

I validated the updated code again on my board. The 'v' token prints the MPU data once and the 'V' token prints the MPU data for every loop. I got the readings changed with my movements.

sskorol, 2022-02-08:
@borntoleave what lib/tool do you use for plotting real-time data?

borntoleave, 2022-04-14:
The 2.0 branch has a switch case for printing gyros. token -v and -V

closed 2022-04-14 by borntoleave