PetoiCamp/OpenCat-Quadruped-Robot #30: Is it possible to remote-control precise Bittle gaits?

slerman12, 2022-08-06:
I've seen examples such as pyBittle of sending pre-programmed gaits to the battle via WiFi or Bluetooth. I'm an AI researcher looking to instead send precise velocities and torques to the Bittle for exact joint control. I also need to capture the Bittle's accelerometer and gyroscope data.

Is there example code anywhere for how to achieve both of these goals?

1. Send precise commands
2. Receive Bittle metadata 

Thanks for any help!

borntoleave, 2022-08-07:
Hi,
1. Bittle's servo moves by position control. You can send interpolated
positions at different time intervals to tune the velocities. Our official
example can be found in serialMaster/example.py.

2. In OpenCat.h, you can uncomment
#define T_PRINT_GYRO            'v' //print Gyro data
#define T_VERBOSELY_PRINT_GYRO  'V' //verbosely print Gyro data
and send 'v' or 'V' to get the IMU data. The current code only sends gyro
data, but you can print the full 6-axis data (including the acceleration)
by modifying the print6Axis() function in OpenCat/src/imu.h.

slerman12, 2022-08-07:
Oof, this may be beyond my expertise.

I'm gonna really need your input if I'm to do this.

Okay, I've looked at `example.py`.

> You can send interpolated positions at different time intervals

1. Is there a line in `example.py`  that sends a "position"? I'm not sure what is meant by this.
2. Is it possible to send positions via Bluetooth or WiFi? There is a library called pyBittle that makes sending commands to Bittle via Bluetooth/WiFi really easy, but they only by default send full pre-programmed gaits.

I've taken a look at `opencat.h` and `imu.h`, but this is where I really don't have the expertise since I don't program outside of Python often.

> you can print the full 6-axis data (including the acceleration) by modifying the print6Axis() function in OpenCat/src/imu.h

1. I see the following code in `imu.h`. How can I modify it to output accelerometer data in addition to gyroscope data? Any hints?

```
void print6Axis() {
  PT(ypr[0] );
  PT('\t');
  PT(ypr[1] );
  PT('\t');
  PT(ypr[2] );
  PTL();
}
```

2. Is it possible to receive these print signals via Bluetooth or WiFi?

Thank you so much!

borntoleave, 2022-08-09:
1. In OpenCat/serialMaster/demos, you can find an example moveBySteps.py that moves specific joints by time intervals. 
2. Yes, you can connect the Bluetooth dongle to your computer and then run any of the examples that import the ardSerial.py. There's an algorithm in the code that automatically detects connected robots. 
3. To get more IMU data, simply uncomment the lines after print ypr[]
```c++
void print6Axis() {
  PT(ypr[0] );
  PT('\t');
  PT(ypr[1] );
  PT('\t');
  PT(ypr[2] );
  PT("\t");
  PT(aaWorld.x);
  PT("\t");
  PT(aaWorld.y);
  PT("\t");
  PTL(aaWorld.z / 1000);
  PT('\t');
  PT(aaReal.z);
  PTL();
}

```

If you send a 'v' token, the angles will be printed to the serial port for once. And the Bluetooth or WiFi dongle can transfer it to your computer. If you send a 'V' token, the angles will be continuously printed to the serial port. 

Currently, the Bluetooth code is fully incorporated in the ardSerial.py.

closed 2022-10-25 by borntoleave

slerman12, 2023-02-09:
Hi Rz, this is Sam. Currently, I'm testing Bluetooth bidirectional commands. Here is the output when I send the following command:

`b'v'`

```
0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific: bytearray(b'G\r\n-21.26759\t10.8250')
0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific: bytearray(b'7\t-11.13954\r\nv\r\n')
```

Here is the output when I send the following command:

`b'k'`

```
0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific: bytearray(b'G\r\n')
0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific: bytearray(b'k\r\n')
```

In this case, Bittle briefly moves his head.

Here is the output when I send the following command:

`b'-9'`

`0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific: bytearray(b'G\r\n-\r\n')`

Bittle beeps, but nothing happens. (Same for other variations).

Can you show me how to send individual commands and bytearrays of commands? I'm using a special library because my M1 MacBook was having issues with the Bluetooth code.

slerman12, 2023-02-10:
This seems to work:

`str('m 0 45 0 -45 0 45 0 -45').encode('utf-8')`

But this doesn't:

`str('l 20 0 0 0 0 0 0 0 45 45 45 45 36 36 36 36').encode('utf-8')`

And neither does this:

`str('m 1 45 2 -45 3 45').encode('utf-8')`

Here is my best solution. Is there a better way than this that doesn't require calling `time.sleep(1)` and can automatically detect when a command has finished executing?

```
commands = list(map(encode, [[45, 0, 0, 0, 0, 0, 0, 0, 45, 45, 45, 45, 36, 36, 36, 36],
                             [-45, 0, 0, 0, 0, 0, 0, 0, 45, 45, 45, 45, 36, 36, 36, 36],
                             [45, 0, 0, 0, 0, 0, 0, 0, 45, 45, 45, 45, 36, 36, 36, 36],
                             [-45, 0, 0, 0, 0, 0, 0, 0, 45, 45, 45, 45, 36, 36, 36, 36]]))

def encode(rotations: list):
    return ('i ' + ' '.join(map(str, [*sum(zip(servos, rotations), ())]))).encode('utf-8')

def send(data):
    ...

for command in commands:
    # Send command via Bluetooth
    send(command)
  
    time.sleep(1)
```

borntoleave, 2023-02-10:
Hi, please download the latest OpenCat
<https://github.com/PetoiCamp/OpenCat> firmware on GitHub and upload it to
your Bittle. I just optimized the overall stability of the serial
communication. You can refer to the serialMaster/example.py for the
posture-related commands and the function to send them. Pay attention to
the brackets and letter cases.

            # - I indicates the command to rotate multiple joint servos at
the same time

            # - 8, 9, 10, 11, 0 indicate the index numbers of joint servos

            # - 50, 50, 50, 50, 0 indicate the rotation angle (this angle
refers to the origin,

            #                     rather than additive), the unit is degree

            # - 3 indicates the postponed time after finishing the command,
in seconds

            ['I', [8, -20, 0, 0, 9, -20, 10, 20, 11, 20, ], 1],

            ['I', [8, 20, 0, 40, 9,  20, 10, 50, 11, 50, ], 1],

            # - L indicates the command to control all joint servos to
rotate at the same time

            #     (currently the command supports 16 degrees of freedom,
that is, 16 servos)

            # - 20,0,0,0,0,0,0,0,45,45,45,45,36,36,36,36 indicate the
rotation

            #               angle of each joint servo corresponding to 0-15
(this angle refers to

            #               the origin, rather than additive), the unit is
degree

            # - 5 indicates the postponed time after finishing the command,
in seconds

            ['L', [20, 0, 0, 0, 0, 0, 0, 0, 5, 0, 45, 45, 80, 80, 36, 36], 1
],

For the 'I' commands, only 5 pairs of indexes and joint is allowed, making
the total length to 10. You may change the limit in OpenCat.h:
#define CMD_LEN 10 //the last char will be '\0' so only CMD_LEN-1 elements
are allowed

If you want to change more joints at once, the 'L' command encodes the
whole 16 joints and is more efficient.

slerman12, 2023-02-12:
Since I'm reading the IMU regularly, it's vital that I disable the beeping. Is it possible to send Bittle a command like `b'v'` soundlessly?

slerman12, 2023-02-12:
Is it possible to get IMU data _while_ Bittle is performing actions?

borntoleave, 2023-02-13:
[image]

slerman12, 2023-02-28:
I need to define constraints on Bittle's movements so that random actions don't cause him to twist a leg into his body or collide with himself. What are some of the red-zone ranges, do you know? Depending on where the other legs are, different movements are acceptable. So I may have to come up with this myself, but do you have an intuition for a general rule of thumb that can be used to constrain his movements depending on his other movements? In my scenario, he starts out taking random actions and this can be too demanding on the build.

borntoleave, 2023-03-05:
There's a simple rule to limit a single joint's range in OpenCat.h. 
```cpp
int angleLimit[][2] = {
  { -120, 120 }, { -30, 80 }, { -120, 120 }, { -120, 120 },
  { -90, 60 }, { -90, 60 }, { -90, 90 }, { -90, 90 },
  { -200, 80 }, { -200, 80 }, { -80, 200 }, { -80, 200 },
  { -80, 200 }, { -80, 200 }, { -80, 200 }, { -80, 200 },
};

You will need to calculate the exact legs' states to evaluate the collision with other body parts. The calculation could be done on the BiBoard or your master computer that sends the movement instructions. It's very challenging to fit that algorithm into NyBoard.

slerman12, 2023-06-23:
Hi Rz, I'm returning to this. Is it alright if I ask some questions that I stumbled into last time?

1. Reproducing the pre-programmed gaits via the kind of position control above seems... there seems to be something missing. I think the pre-programmed gaits have different velocities, right? I want to give the robot an action space from which it could feasibly learn to reproduce the pre-programmed gaits.
2. Are there any high-level examples of miniature movements like taking just _one_ step forward, or _one_ step to the right, etc.?
3. is it possible to disable the automatic trigger movements, like flipping upright when Bittle collapses. It's precisely this kind of movement that I have a hard time imagining the above action space reproducing.

Thank you.

borntoleave, 2023-06-24:
Hi,
Have you tried the SkillComposer
<https://docs.petoi.com/desktop-app/skill-composer>, which allows you to
import the entire instinct library and inspect each skill frame? You can
define the gait as a behavior and change the speed and loop cycles.
A skill data array can also be sent through the serial API
<https://docs.petoi.com/apis/python-api> so that you don't need to modify
and re-upload the entire firmware.
You can send capitalized 'G' as a serial command field to disable the gyro
reaction.

slerman12, 2023-06-28:
I haven't tried the SkillComposer yet. It confuses me. I'm looking to do everything in Python.

Is it possible to send velocities through the serial API? It would be really helpful to see some examples of complex behaviors like flipping upright or running done with the serial API commands listed in the link you sent me.

Thanks,
Sam

borntoleave, 2023-06-29:
The skill composer can allow you to inspect the keyframes one by one and
test them on the go. The Python API can use the Skill array exported from
SkillComposer.
In serialMaster/example.py, you can see how the skill is sent to the robot.
The array "ck" is a behavior in the same format as flipping. You can start
the command with the capital 'K'.

slerman12, 2023-08-08:
Is there a simple way to read/verify that Bittle is (a) standing upright and (b) has a positive forward velocity? I retrieve 6 values from the IMU, 3 from the gyroscope and 3 from the accelerometer.

borntoleave, 2023-08-10:
Hi, you can use the token v or V to retrieve the 6-axis IMU data. Integrating the acceleration can give you speed information.

borntoleave, 2023-08-13:
The 6-axis data are defined in OpenCat/src/imu.h.
In print6Axis() prints out yaw, pitch, roll in angles, and acceleration in x, y, z directions. 

The sensor can only detect acceleration rather than the current velocity. The velocity is acquired by integrating the acceleration in a specific direction. You may lose the initial speed, and the error may accumulate. Because a quadruped always wobbles and shakes during motion, you can assume that after the acceleration in all directions is stable for one second, the robot is still. Then you can use it as the integration starting point.