PetoiCamp/OpenCat-Quadruped-Robot #31: Application crashes when using MacOS with M1

ioulosve, 2022-08-29:
I'm experiencing issues launching the Petoi Desktop App with a Macbook Pro with M1 CPU.

Launching from the Launchpad I only got this error message:

[image]

Launching from the terminal I got this traceback:

`
Traceback (most recent call last):
  File "/Applications/Petoi Desktop App.app/Contents/Resources/__boot__.py", line 89, in _recipes_pil_prescript
    import Image
ModuleNotFoundError: No module named 'Image'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Applications/Petoi Desktop App.app/Contents/Resources/__boot__.py", line 135, in <module>
    _recipes_pil_prescript(['DdsImagePlugin', 'FitsStubImagePlugin', 'EpsImagePlugin', 'WmfImagePlugin', 'PdfImagePlugin', 'JpegImagePlugin', 'DcxImagePlugin', 'GbrImagePlugin', 'FpxImagePlugin', 'PalmImagePlugin', 'IptcImagePlugin', 'PsdImagePlugin', 'SunImagePlugin', 'MpegImagePlugin', 'IcoImagePlugin', 'BmpImagePlugin', 'IcnsImagePlugin', 'TgaImagePlugin', 'GifImagePlugin', 'FliImagePlugin', 'Jpeg2KImagePlugin', 'SgiImagePlugin', 'Hdf5StubImagePlugin', 'CurImagePlugin', 'PixarImagePlugin', 'BufrStubImagePlugin', 'XpmImagePlugin', 'MicImagePlugin', 'PngImagePlugin', 'BlpImagePlugin', 'WebPImagePlugin', 'PcdImagePlugin', 'FtexImagePlugin', 'MpoImagePlugin', 'MspImagePlugin', 'ImtImagePlugin', 'PpmImagePlugin', 'SpiderImagePlugin', 'PcxImagePlugin', 'GribStubImagePlugin', 'McIdasImagePlugin', 'XbmImagePlugin', 'ImImagePlugin', 'XVThumbImagePlugin', 'TiffImagePlugin'])
  File "/Applications/Petoi Desktop App.app/Contents/Resources/__boot__.py", line 93, in _recipes_pil_prescript
    from PIL import Image
  File "<frozen zipimport>", line 259, in load_module
  File "PIL/Image.pyc", line 89, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "PIL/_imaging.pyc", line 14, in <module>
  File "PIL/_imaging.pyc", line 10, in __load
  File "imp.pyc", line 342, in load_dynamic
ImportError: dlopen(/Applications/Petoi Desktop App.app/Contents/Resources/lib/python3.9/lib-dynload/PIL/_imaging.so, 0x0002): Library not loaded: '@rpath/libtiff.5.dylib'
  Referenced from: '/Applications/Petoi Desktop App.app/Contents/Resources/lib/python3.9/lib-dynload/PIL/_imaging.so'
  Reason: tried: '/opt/anaconda3/envs/simpleUI/lib/libtiff.5.dylib' (no such file), '/opt/anaconda3/envs/simpleUI/lib/libtiff.5.dylib' (no such file), '/Applications/Petoi Desktop App.app/Contents/Resources/lib/python3.9/lib-dynload/PIL/../../../libtiff.5.dylib' (no such file), '/opt/anaconda3/envs/simpleUI/lib/libtiff.5.dylib' (no such file), '/opt/anaconda3/envs/simpleUI/lib/libtiff.5.dylib' (no such file), '/Applications/Petoi Desktop App.app/Contents/Resources/lib/python3.9/lib-dynload/PIL/../../../libtiff.5.dylib' (no such file), '/opt/anaconda3/envs/simpleUI/lib/libtiff.5.dylib' (no such file), '/Applications/Petoi Desktop App.app/Contents/Frameworks/libtiff.5.dylib' (no such file), '/Applications/Petoi Desktop App.app/Contents/MacOS/../Frameworks/libtiff.5.dylib' (no such file), '/usr/local/lib/libtiff.5.dylib' (no such file), '/usr/lib/libtiff.5.dylib' (no such file)
2022-08-29 10:16:05.365 Petoi Desktop App[7205:38695] Launch error
2022-08-29 10:16:05.365 Petoi Desktop App[7205:38695] Launch error
See the py2app website for debugging launch issues
exit
`

I also tried directly from source code cloning this repo and starting pyUI/UI.py after have installed pyserial and pillow packages and the application starts showing the small window with the three buttons but then, when I choose Skill Composer it crashes with a Segmentation fault error when I click con Behavior or Export buttons.

Is there something I'm missing?

Thanks in advance for the help

borntoleave, 2022-08-30:
> File "", line 259, in load_module
> 0x0002): Library not loaded: ***@***.***/libtiff.5.dylib'

I developed the code on an M1 mac, so there shouldn't be a compatibility
issue.
Since you have tried to install pillow, but the program still reports a "No
module named 'Image'" error, I suggest you check if your python library
path and the running path are the same.

ioulosve, 2022-08-31:
Checking the traceback I noticed that the App searches for `/opt/anaconda3/envs/simpleUI/lib/libtiff.5.dylib` library so I installed anaconda (I was using minicoda before) and then I created a enviroment called simpleUI where I installed pillow and pyserial. 

Now the Petoi Desktop App found that library but it says that the architecture is incompatible (have arm64, need x86_64).

What I'm supposed to do now?

```
Traceback (most recent call last):
  File "/Applications/Petoi Desktop App.app/Contents/Resources/__boot__.py", line 89, in _recipes_pil_prescript
    import Image
ModuleNotFoundError: No module named 'Image'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Applications/Petoi Desktop App.app/Contents/Resources/__boot__.py", line 135, in <module>
    _recipes_pil_prescript(['DdsImagePlugin', 'FitsStubImagePlugin', 'EpsImagePlugin', 'WmfImagePlugin', 'PdfImagePlugin', 'JpegImagePlugin', 'DcxImagePlugin', 'GbrImagePlugin', 'FpxImagePlugin', 'PalmImagePlugin', 'IptcImagePlugin', 'PsdImagePlugin', 'SunImagePlugin', 'MpegImagePlugin', 'IcoImagePlugin', 'BmpImagePlugin', 'IcnsImagePlugin', 'TgaImagePlugin', 'GifImagePlugin', 'FliImagePlugin', 'Jpeg2KImagePlugin', 'SgiImagePlugin', 'Hdf5StubImagePlugin', 'CurImagePlugin', 'PixarImagePlugin', 'BufrStubImagePlugin', 'XpmImagePlugin', 'MicImagePlugin', 'PngImagePlugin', 'BlpImagePlugin', 'WebPImagePlugin', 'PcdImagePlugin', 'FtexImagePlugin', 'MpoImagePlugin', 'MspImagePlugin', 'ImtImagePlugin', 'PpmImagePlugin', 'SpiderImagePlugin', 'PcxImagePlugin', 'GribStubImagePlugin', 'McIdasImagePlugin', 'XbmImagePlugin', 'ImImagePlugin', 'XVThumbImagePlugin', 'TiffImagePlugin'])
  File "/Applications/Petoi Desktop App.app/Contents/Resources/__boot__.py", line 93, in _recipes_pil_prescript
    from PIL import Image
  File "<frozen zipimport>", line 259, in load_module
  File "PIL/Image.pyc", line 89, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "PIL/_imaging.pyc", line 14, in <module>
  File "PIL/_imaging.pyc", line 10, in __load
  File "imp.pyc", line 342, in load_dynamic
ImportError: dlopen(/Applications/Petoi Desktop App.app/Contents/Resources/lib/python3.9/lib-dynload/PIL/_imaging.so, 0x0002): Library not loaded: '@rpath/libtiff.5.dylib'
  Referenced from: '/Applications/Petoi Desktop App.app/Contents/Resources/lib/python3.9/lib-dynload/PIL/_imaging.so'
  Reason: tried: '/opt/anaconda3/envs/simpleUI/lib/libtiff.5.dylib' (mach-o file, but is an incompatible architecture (have (arm64), need (x86_64))), '/opt/anaconda3/envs/simpleUI/lib/libtiff.5.dylib' (mach-o file, but is an incompatible architecture (have (arm64), need (x86_64))), '/Applications/Petoi Desktop App.app/Contents/Resources/lib/python3.9/lib-dynload/PIL/../../../libtiff.5.dylib' (no such file), '/opt/anaconda3/envs/simpleUI/lib/libtiff.5.dylib' (mach-o file, but is an incompatible architecture (have (arm64), need (x86_64))), '/opt/anaconda3/envs/simpleUI/lib/libtiff.5.dylib' (mach-o file, but is an incompatible architecture (have (arm64), need (x86_64))), '/Applications/Petoi Desktop App.app/Contents/Resources/lib/python3.9/lib-dynload/PIL/../../../libtiff.5.dylib' (no such file), '/opt/anaconda3/envs/simpleUI/lib/libtiff.5.dylib' (mach-o file, but is an incompatible architecture (have (arm64), need (x86_64))), '/Applications/Petoi Desktop App.app/Contents/Frameworks/libtiff.5.dylib' (no such file), '/Applications/Petoi Desktop App.app/Contents/MacOS/../Frameworks/libtiff.5.dylib' (no such file), '/usr/local/lib/libtiff.5.dylib' (no such file), '/usr/lib/libtiff.5.dylib' (no such file)
2022-08-31 17:25:43.758 Petoi Desktop App[11550:81866] Launch error
2022-08-31 17:25:43.758 Petoi Desktop App[11550:81866] Launch error
See the py2app website for debugging launch issues
```

ioulosve, 2022-09-02:
Ok I solved it installing the x86_64 version of anaconda3 in /opt/anaconda3, creating the simpleUI environment and installing the pyserial and pillow packages inside it.

I hope these information can help someone with the same problem

Thank you for the support

closed 2022-09-02 by ioulosve