PetoiCamp/OpenCat-Quadruped-Robot #32: Is this a bug? I can't understand the logic here

yaringal, 2022-08-29:
https://github.com/PetoiCamp/OpenCat/blob/1abeb1cf15f4dda2b9f16e23810eb3a9e0289fd1/src/motion.h#L65

borntoleave, 2022-08-30:
for (i % 4 < 2) ? ( RollPitchDeviation[1]) : RollPitchDeviation[1]

It's not a bug but a useless branch condition. Previously I tested two
cases but then deleted the factors in the first case. That's why the first
RollPitchDeviation[1] has parentheses around it. I left it there in case I
want to experiment more in the future.

borntoleave, 2022-08-30:
> many thanks for the quick reply! To make sure I understand the logic in
> this function - is the following interpretation correct? Whenever the IMU
> error is not negligible, the actuators will decrease the leg angles to get
> the centre of mass closer to the ground. The expected body orientation in
> the skills is only used to calculate the error, but the robot will not try
> to attain the body orientation specified. Is this correct?

The expected body orientation is a reference angle used to tell how much
the robot is off the right position. The error is fed into a simple
feedback calculation. My code only tries to reduce the error, but not to
eliminate it. Ideally, if you design a good PID control, you may attain the
specified expected body orientation.

closed 2022-10-25 by borntoleave