PetoiCamp/OpenCat-Quadruped-Robot #38: MU Camera using arduino IDE error

Getafix17, 2023-02-27:
I just got a  camera module, when I uncomment  // [#define](https://www.youtube.com/hashtag/define) CAMERA   //for BallTracking using Mu Vision camera I get the following error in Arduino IDE 
Arduino: 1.8.19 (Windows 10), Board: "Arduino Uno"
In file included from C:\Users\alexf\Desktop\Petoi Bittle\Bittle OpenCat\OpenCat\src\OpenCat.h:443:0,

                 from C:\Users\alexf\Desktop\Petoi Bittle\Bittle OpenCat\OpenCat\OpenCat.ino:81:

C:\Users\alexf\Desktop\Petoi Bittle\Bittle OpenCat\OpenCat\src\reaction.h: In function 'void reaction()':

C:\Users\alexf\Desktop\Petoi Bittle\Bittle OpenCat\OpenCat\src\reaction.h:340:24: error: expected primary-expression before ':' token

       case T_TASK_QUEUE:

                        ^

exit status 1

Error compiling for board Arduino Uno.
I tried using the bug fix recommended by Petoi camp In src/opencat.h, find #define T_TASK_QUEUE, add space and ‘q’ after it. 
#define T_TASK_QUEUE ‘q’ did not work either.

borntoleave, 2023-02-27:
Try using [this version](https://github.com/PetoiCamp/OpenCat/tree/beforeInplace) in the working branch. It will be merged into the main after a thorough test.

Getafix17, 2023-02-27:
Thanks that worked, I was able to compile the code without error after uncommenting #define CAMERA //

closed 2023-02-27 by Getafix17