PetoiCamp/OpenCat-Quadruped-Robot #41: very-long-line-length-in-source-file  > 512

atzlinux, 2023-03-09:
When I make deb package for this repo, Debian check tool find the some files's line is very long.

I suggest to modify them:

very-long-line-length-in-source-file 9290 > 512 [ModuleTests/ESP8266WiFiController/mainpage.h:7]
very-long-line-length-in-source-file 782 > 512 [README.md:13]
very-long-line-length-in-source-file 602 > 512 [ModuleTests/ESP8266WiFiController/calibrationpage.h:9]

Thanks!

atzlinux, 2023-03-10:
> They are the Arduino sample codes to render a webpage. They are not
> included in the desktop app. Do we still need to shorten them?

This is the Debian's explanation about this check:

N:
P: very-long-line-length-in-source-file
N:
N:   The source file includes a line length that is well beyond the normally
N:   human made code line length.
N:
N:   This very long line length does not allow Lintian to do correctly some
N:   source file checks.
N:
N:   This line could also be the result of some text injected by a computer
N:   program, and thus could lead to FTBFS bugs.
N:
N:   Last but not least, long line in source code could be used to obfuscate
N:   the source code and to hide stuff like backdoors or security problems.
N:
N:   It could be due to jslint source comments or other build tool comments.
N:
N:   You may report this issue upstream.

Certainly, this is only a suggestion for deb packaging.

The README.md  file is included in the desktop app deb package as a user 
document.

borntoleave, 2023-03-10:
The extremely long string in mainpage.h is the base64 format of our logo
image.
The Readme has been updated to shorter sentences.
The calibration page has a long paragraph section. It could be split.

closed 2023-03-16 by borntoleave