PetoiCamp/OpenCat-Quadruped-Robot #42: Unable to upload code after bluetooth connection failed

Getafix17, 2023-03-11:
After I tried to upload the new main sketch it failed to connect to bluetooth. Now for some reason I cannot  upload anything with the Arduino IDE or desktop apk.

Error from Arduino IDE 

Arduino: 1.8.19 (Windows 10), Board: "Arduino Uno"

Sketch uses 24486 bytes (75%) of program storage space. Maximum is 32256 bytes.

Global variables use 1471 bytes (71%) of dynamic memory, leaving 577 bytes for local variables. Maximum is 2048 bytes.

An error occurred while uploading the sketch

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x7d

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x7d

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x7d

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x7d

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x7d

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x7d

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x7d

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x7d

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x7d

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x7d

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

borntoleave, 2023-03-11:
Try to upload the sketch using the USB uploader. It's more stable than the
Bluetooth dongle.
When using the USB uploader, observe its LEDs. Is the yellow LED constantly
lighting up or does it blink after you press the upload button?

Getafix17, 2023-03-11:
I am using the USB uploader. The yellow LED is constant not blinking, Also the beeping sounds I used to get while connecting the robot have stopped and the robot is completly dead.

Getafix17, 2023-03-11:
Desktop apk also gives an upload error

borntoleave, 2023-03-11:
The Arduino bootloader may be crashed. Please check this section to reset:
[Burn the bootloader](https://docs.petoi.com/arduino-ide/upload-sketch-for-nyboard#burn-the-bootloader-no-need-for-normal-use)

Getafix17, 2023-03-11:
Thanks Dr. Li. I will try to reset the bootloader. I will close for now. 
P.S. I am a big fan of your work!

Getafix17, 2023-03-11:
I have an Arduino UNO. According to the documentation, "The programming process uses VCC, GND and four data pins. Three pins connect MISO (pin12), MOSI(Pin11) and SCK(pin13) between the programming micro and the target micro, the fourth pin from the programming micro goes to the reset pin of the target."
The serial USB adapter has GND,RTS, RX, TX, 5V,GND, DTR. How do I connect these to the corresponding arduino pins using jumper wires. Also how could I know if it is a hardware problem?

borntoleave, 2023-03-12:
I just added a pinout picture in the user manual. Please check. 
[image]

Getafix17, 2023-03-12:
Thank you sir, I have one more question, please forgive my ignorance. Can I connect the jumper wires directly from the top of the board. I used jumper wires and the contact seems loose.

borntoleave, 2023-03-12:
You can press the pins slightly to ensure the connections. It's a onetime operation, so no need for soldering.

Getafix17, 2023-03-12:
Thanks for your patience sir. I will try it out.

closed 2023-03-12 by Getafix17