PetoiCamp/OpenCat-Quadruped-Robot #64: After uploading the code of OpenCat.ino Robot will not accept any new code.

JEAPI-DEV, 2023-11-02:
So I wanted to upload the OpenCat.ino Program for testing, and it worked but after doing so it's Buzzer piped and after that I was not able to upload any new code to the Bittle Bot. Be it over https://ide.tinkergen.com/ or via the Arduino IDE. 

So then I tried uploading the factory code explained in the documentation. But this is the error I got: 

`C:\Users\jeapi\AppData\Local\Programs\cc-assistant\resources\script\arduino\opencat>"C:\Users\jeapi\AppData\Local\Programs\cc-assistant\resources\compilers\arduino"\contents\hardware\tools\avr\bin\avrdude -C"C:\Users\jeapi\AppData\Local\Programs\cc-assistant\resources\compilers\arduino"\contents\hardware\tools\avr\etc\avrdude.conf -v -patmega328p -carduino -P"COM10" -b115200 -D -Uflash:w:"C:\Users\jeapi\AppData\Local\Programs\cc-assistant\resources\buildTemp\arduino\opencat\grove_.ino.hex":i C:\Users\jeapi\AppData\Local\Programs\cc-assistant\resources\script\arduino\opencat>del "C:\Users\jeapi\AppData\Local\Programs\cc-assistant\resources\buildTemp\arduino\opencat\grove_.ino.hex" avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53 Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/ Copyright (c) 2007-2014 Joerg Wunsch System wide configuration file is "C:\Users\jeapi\AppData\Local\Programs\cc-assistant\resources\compilers\arduino\contents\hardware\tools\avr\etc\avrdude.conf" Using Port : COM10 Using Programmer : arduino Overriding Baud Rate : 115200 avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xbd avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xbd avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xbd avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xbd avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xbd avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xbd avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xbd avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xbd avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xbd avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xbd avrdude done. Thank you.
`

Thanks in advance.

JEAPI-DEV, 2023-11-02:
The problem seems to be So the problem seems to be that I flashed this code, but forgot to chnage the NyBoard to NvBoard_V1_1. It was set by default to NyBoard_V1_2 .

borntoleave, 2023-11-02:
> code, but forgot to change the NyBoard to NvBoard_V1_1. It was set by

The board version definition shouldn't block new uploads.
Please follow this debugging routine:
https://docs.petoi.com/technical-support/faq-frequently-asked-questions#nyboard-cant-upload-firmware

closed 2023-11-02 by JEAPI-DEV