PetoiCamp/OpenCat-Quadruped-Robot #65: Sketch too big for NyBoard V0_2

wack-back, 2023-11-07:
When compile with #define MAIN_SKETCH uncomment, the arduino gives:
Sketch uses 31824 bytes (103%) of program storage space. Maximum is 30720 bytes.
Global variables use 1596 bytes (77%) of dynamic memory, leaving 452 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Any way can solve this?

borntoleave, 2023-11-08:
NyBoard V0_* used a 20MHz crystal faster than standard Uno's 16MHz. Its bootloader is borrowed from Arduino Pro mini and takes more space. We've added new functions to the original code until it filled up NyBoard V1_* and exceeded NyBoardV0_*.

The mobile app has already replaced the infrared remote's functionality. You may turn off the infrared remote module by commenting out this line at the beginning of OpenCat.h. The size will be reduced to 99%. 
`// #define IR_PIN 4  // Signal Pin of IR receiver to Arduino Digital Pin 4`
[image]

Alternatively, you may try our new boards:

- [Uno-equivalent NyBoard V1](https://www.petoi.com/products/nyboard-customized-arduino-board)

- [ESP32-based BiBoard](https://www.petoi.com/products/biboard-esp32-development-board-for-quadruped-robot)

wack-back, 2023-11-16:
Thank you for solving this issue!

closed 2023-11-16 by borntoleave