PetoiCamp/OpenCat-Quadruped-Robot #67: Petoi Bluetooth Adapter not compatible with NyBoard_V02 ?

wack-back, 2023-11-16:
I bought Petoi Bluetooth Adapter, and plug it into the NyBoard_V02.
However, the mobile app seems not compatible with it.
I tried both Apps in https://docs.petoi.com/mobile-app/introduction
I can see the PetoiBLE-9ACC listed in Petoi app. But when I click on link button, the NyBoard restarts, and App shows can't find Petoi device.
Anyway I can solve this? or I can only use Petoi Bluetooth Adapter as a remote uart?

wack-back, 2023-11-17:
Issue solved! It's because the NyBoard_V02 baudrate default to 57600, but BLE is 115200 as default.
Just change the baudrate in OpenCat.h
Problem Solved!

closed 2023-11-17 by wack-back

borntoleave, 2023-11-17:
> Issue solved! The NyBoard_V02 baudrate defaults to 57600, but BLE is 115200 as default. Just change the baudrate in OpenCat.h Problem Solved!

Ah, yes, It's been quite a while, and I just forgot to mention the difference in Baudrate. Glad that you pointed it out.