PetoiCamp/OpenCat-Quadruped-Robot #70: pyUI/commonVar.py uses wrong paths for Linux

nrwahl2, 2024-01-01:
```
elif platform.system() == "Linux":    # for Linux
    sys.path.append('/usr/share/petoi-opencat/serialMaster/')
    resourcePath = '/usr/share/petoi-opencat/resources/'
    releasePath = '/usr/share/petoi-opencat/release/'
```

There's no reason these directories would be subdirectories of `/usr/share/petoi-opencat`. There's no installation procedure for Linux that would put them there. The Petoi documentation says to run `python3 UI.py` from a source checkout. There's no `make install` and there's no package (RPM, Pip, or otherwise).

JasonWong08, 2024-01-01:
Thanks for your feedback. These scripts are used for the Linux user who has installed the desktop app before. We will fix this issue in the next release soon.

nrwahl2, 2024-01-01:
Thanks @JasonWong08. I may have overlooked it, or it might be legacy -- I didn't see a procedure for installing the desktop app on Linux, that would install files in standard directories like `/usr/share`.

closed 2024-02-05 by borntoleave