PetoiCamp/OpenCat-Quadruped-Robot #77: Getting data from IMU data (such as YRP and Accelerators) on the main source

Kakcalu13, 2024-02-15:
https://github.com/PetoiCamp/OpenCat/blob/main/OpenCat.ino

I'm using above as a main source. I don't see any IMU data. I tried to do the same thing from this file, https://github.com/PetoiCamp/OpenCat/tree/main/ModuleTests/testMPU on the main source and it end up insufficient memory to upload. 

What do you suggest? No matter what I do, I break everything when I tried something in the main source

Kakcalu13, 2024-02-17:
i was able to print after ran a command 'V' but it only displays 2 things instead of 3. I also wanted to display acceleration + gyro in real time

borntoleave, 2024-02-19:
In src/imu.h, in print6Axis(), you can enable the line for the yaw
(ypr[0]). YPR means yaw pitch roll, corresponding to 0,1,2.
If the flash runs low, you can comment out the *#define IR_PIN* line to
save some space.

Kakcalu13, 2024-02-19:
those worked!! thank you so much! What about the acceleration?

borntoleave, 2024-02-20:
You can refer to OpenCat/ModuleTests/testMPU/testMPU.ino and activate the
defined macros to print the accelerations. Then add those statements to the
OpenCat code.

Kakcalu13, 2024-02-20:
Hi there,

I truly appreciate your time! 

I tried that and the memory end up maxed. See here:

[image]

All I did was update this in the imu.h

```
// #define READ_ACCELERATION
void print6Axis() {
  PT_FMT(ypr[0], 5);
  PT('\t');
  PT_FMT(ypr[1], 5);
  PT('\t');
  PT_FMT(ypr[2], 5);
  PT("\t");
  mpu.dmpGetQuaternion(&q, fifoBuffer);
  mpu.dmpGetAccel(&aa, fifoBuffer);
  mpu.dmpGetGravity(&gravity, &q);
  mpu.dmpGetLinearAccel(&aaReal, &aa, &gravity);
  mpu.dmpGetLinearAccelInWorld(&aaWorld, &aaReal, &q);
  Serial.print("aworld\t");
  Serial.print(aaWorld.x);
  Serial.print("\t");
  Serial.print(aaWorld.y);
  Serial.print("\t");
  Serial.println(aaWorld.z);
```

I even tried to do it on OpenCat.h:

```
#ifdef GYRO_PIN
  for (byte r = 0; r < 50; r++) {  //ypr is slow when starting up. leave enough time between IMU initialization and this reading
    read_IMU();
    mpu.dmpGetQuaternion(&q, fifoBuffer);
    mpu.dmpGetAccel(&aa, fifoBuffer);
    mpu.dmpGetGravity(&gravity, &q);
    mpu.dmpGetLinearAccel(&aaReal, &aa, &gravity);
    mpu.dmpGetLinearAccelInWorld(&aaWorld, &aaReal, &q);
    Serial.print("aworld\t");
    Serial.print(aaWorld.x);
    Serial.print("\t");
    Serial.print(aaWorld.y);
    Serial.print("\t");
    Serial.println(aaWorld.z);
    delay(10);
  }
  token = (fabs(ypr[1]) > 30 || fabs(ypr[2]) > 30) ? T_CALIBRATE : T_REST;  //put the robot's side on the table to enter calibration posture for attaching legs
  newCmdIdx = 2;
  int rndSeed = ypr[2] * 10000;
#else
  int rndSeed = analogRead(A2);
#endif
```

I must be doing something wrong. Can you tell me which is the macros? Also do you happen to have a doc to show where or how?

borntoleave, 2024-02-21:
By disabling this IR_PIN, you can save 4% of the flash.
[image]

[image]

Kakcalu13, 2024-02-21:
Thank you for your time again!

Yeah, it's already disabled.
[image]

I was just confused about how this code snippet ended up consuming more memory than IR when it's just 11 lines:

```cpp
  mpu.dmpGetQuaternion(&q, fifoBuffer);
  mpu.dmpGetAccel(&aa, fifoBuffer);
  mpu.dmpGetGravity(&gravity, &q);
  mpu.dmpGetLinearAccel(&aaReal, &aa, &gravity);
  mpu.dmpGetLinearAccelInWorld(&aaWorld, &aaReal, &q);
  Serial.print("aworld\t");
  Serial.print(aaWorld.x);
  Serial.print("\t");
  Serial.print(aaWorld.y);
  Serial.print("\t");
  Serial.println(aaWorld.z);
```

See here in `opencat.h`:

```cpp
#ifdef GYRO_PIN
  for (byte r = 0; r < 50; r++) {  //ypr is slow when starting up. leave enough time between IMU initialization and this reading
    read_IMU();

// Add new section from testIMU code.
    mpu.dmpGetQuaternion(&q, fifoBuffer);
    mpu.dmpGetAccel(&aa, fifoBuffer);
    mpu.dmpGetGravity(&gravity, &q);
    mpu.dmpGetLinearAccel(&aaReal, &aa, &gravity);
    mpu.dmpGetLinearAccelInWorld(&aaWorld, &aaReal, &q);
    Serial.print("aworld\t");
    Serial.print(aaWorld.x);
    Serial.print("\t");
    Serial.print(aaWorld.y);
    Serial.print("\t");
    Serial.println(aaWorld.z);
// End of the new code inserted above

    delay(10);
  }
  token = (fabs(ypr[1]) > 30 || fabs(ypr[2]) > 30) ? T_CALIBRATE : T_REST;  //put the robot's side on the table to enter calibration posture for attaching legs
  newCmdIdx = 2;
  int rndSeed = ypr[2] * 10000;
#else
  int rndSeed = analogRead(A2);
#endif
```

So, based on your response, it sounds like I did the right thing with the macro part. I saw that it saved 4% of flash!

borntoleave, 2024-02-22:
The MPU calls a heavy library that uses many parameters and algorithms to calculate.

Kakcalu13, 2024-03-04:
Thanks! Truly appreciate your time! 

Closing this issue since we have forum now and it's more active and most people go there so I'm going to be there so people can search for the similar question over there

closed 2024-03-04 by Kakcalu13