PetoiCamp/OpenCat-Quadruped-Robot #93: Use gyro and ultrasonic eyes

josephburnett, 2024-09-30:
Currently the software cannot use the gyro and ultrasonic eyes at the same time. 

https://github.com/PetoiCamp/OpenCat/blob/902c3192224833202c429f59df9678b60d838989/OpenCat.ino#L55

Is this a software limitation (making the compiled code smaller) or a hardware limitation (not enough registers)? I am interested in improving the code to have both working at the same time.

borntoleave, 2024-10-01:
ATmega328P only has 32 KB flash and 2 KB SRAM. The added features and UI eventually filled up the programming space (Flash), making it difficult to run two large modules simultaneously. The 64Kb I2C EEPROM used for saving behaviors data is also filled up. You can comment out the IR_PIN and disable the Neopixels in the Ultrasonic sensor to save space for the Gyro. 

We are moving to [OpenCatESP32](https://github.com/PetoiCamp/OpenCatEsp32), where the computational resources are more abundant.

borntoleave, 2025-03-01:
The newer version that runs on ESP32 allows the use of gyro and ultrasonic eyes simultaneously. 
https://github.com/PetoiCamp/OpenCatEsp32/

closed 2025-03-01 by borntoleave