PetoiCamp/OpenCat-Quadruped-Robot #107: Ability to read IMU data with v1.2.4

abpolym, 2025-04-08:
Hey! I just saw that you released [a new version](https://github.com/PetoiCamp/OpenCat/releases/tag/1.2.4) which contained the following statement for the firmware:

* Allow reading IMU data via the serial protocol

I would like to read all IMU data (accelerometer and gyroscope data in all three directions).
How can I do that?

From what I understood, there are two functions defined in [imu.h](https://github.com/PetoiCamp/OpenCat/blob/main/src/imu.h) that are called in [reaction.h](https://github.com/PetoiCamp/OpenCat/blob/main/src/reaction.h):

* [print6Axis](https://github.com/PetoiCamp/OpenCat/blob/main/src/imu.h#L168-L188) is [triggered](https://github.com/PetoiCamp/OpenCat/blob/main/src/reaction.h#L136) when sending a [v or V token](https://github.com/PetoiCamp/OpenCat/blob/main/src/OpenCat.h#L235-L236). Since `READ_ACCELERATION` is not defined per default and printing the first index of `ypr` (yaw pitch roll) is commented out, this would return only 2 out of 6 possible values: pitch and roll.
* [read_IMU](https://github.com/PetoiCamp/OpenCat/blob/main/src/imu.h#L189-L230) is triggered in multiple instances; 1) whenever readEnvironment (such as in the [driver's loop function](https://github.com/PetoiCamp/OpenCat/blob/main/OpenCat.ino#L104)), 2) [when the robot is initialized](https://github.com/PetoiCamp/OpenCat/blob/main/src/OpenCat.h#L516C8-L526), and 3) in [the perform function of src/skill.h](https://github.com/PetoiCamp/OpenCat/blob/main/src/skill.h#L409-L430) which is triggered whenever a skill (triggered by the [`k` and possibly `K` tokens in OpenCat.h](https://github.com/PetoiCamp/OpenCat/blob/main/src/OpenCat.h#L226-L227)) is being performed? All instances require `GYRO_PIN` to be defined, which is the case by default. However, I do not immediately see how to then retrieve all 6 gyro/acc values.

Is there a way to retrieve acceleration/orientation in all three directions by default or do I have to modify, recompile, and upload the modified firmware?

borntoleave, 2025-04-10:
Hi, 
We have modified the print gyro token to gP, at the bottom of the table:
https://docs.petoi.com/apis/serial-protocol

You can modify the printGyroQ = true to print the values by default. 
[image]

borntoleave, 2025-04-10:
> Hi @borntoleave <https://github.com/borntoleave>
>
> thank you for your answer.
> Do I understand it correctly that this will only print gyro data (pitch
> and roll to be exact) and not accelerometer data?

It will print 6-axis data. Oh, BTW, only the ESP32-based board will print
all 6 data. The NyBoard version must modify the code because the default
firmware has used up all the flash for the generic UI.